Farama-Foundation/Metaworld #372: Replacing mujoco-py with pybind11-based mujoco bindings

ottofabian, 2022-06-27:
Recently, other big benchmark suites, such as [OpenAI gym](https://github.com/openai/gym/blob/024b0f5160716b01effc27de45615bac7620e8b6/README.md?plain=1#L51-L52) and [DMC](https://github.com/deepmind/dm_control/blob/41d0c7383153f9ca6c12f8e865ef5e73a98759bd/README.md?plain=1#L61-L68), switched to the new [pybind11-based mujoco bindings](https://pypi.org/project/mujoco/). 
Since these bindings are made available in DeepMind's official mujoco repository, I expect that they will be properly maintained for quite a while. In addition, this makes installation easier for users as well as potentially allows for a larger user base and fewer issues related to the mujoco installation.

As MetaWorld uses its own [MujocoEnv](https://github.com/rlworkgroup/metaworld/blob/18118a28c06893da0f363786696cc792457b062b/metaworld/envs/mujoco/mujoco_env.py#L31) version,  I suggest including something similar to the [OpenAI gym repo](https://github.com/openai/gym/blob/51a6386a6bdaf3fd0a513133a1cd45ef27735090/gym/envs/mujoco/mujoco_env.py#L56-L74) to allow for a smooth transition and reproducibility of existing results.
While I don't think this change is urgent, I think it is in the long term interest of both users and maintainers.

YouJiacheng, 2022-08-24:
Pros:
+ `mujoco-py` is unmaintained according to https://github.com/deepmind/mujoco#third-party-bindings
+ A copy of the MuJoCo library is provided as part of the package and does not need to be downloaded or installed separately.
Cons:
+ `mujoco` doesn't manage GLContext during rendering, it only provides GLContext API

avnishn, 2022-08-24:
I think I speak for all the maintainers when saying we definitely would love to do this, but we don't have the time. 

If there is anyone who is inclined to do the update themselves that would be great!

YouJiacheng, 2022-08-24:
I can do the update, but I don't want to keep full compatibility with old API which requires a lot of boilerplate codes, especially given that `gym` recently removed the old render API.

ethanluoyc, 2022-11-22:
Is there any update on doing this?

I made an attempt to update the bindings. Right now all scripted policy tests pass, but I am not entirely confident that the update may change the behavior for some environments.

I am also inclined to drop compatibility with the old gym API in favor of new ones, but I am not sure what the decision from the maintainer is. I personally think we should consider just dropping support for old mujoco_py support (i.e., deprecate v1 and migrate v2 in-place or create v3 that uses the new mujoco bindings) as it would bring a lot of maintenance burden both for now and future. If people want to use the old environments they can also check out an old version of the repo.

reginald-mclean, 2023-02-01:
As Meta-World ownership has been transferred from the RLWorkGroup to [Farama Foundation](https://farama.org/), I am closing this issue. This task is on our road map for Meta-World. Please join our [Discord](https://discord.gg/PfR7a79FpQ) to stay up to date on this!

closed 2023-02-01 by reginald-mclean