Farama-Foundation/Metaworld #376: Error when setting rand_init to False in some environments

zchuning, 2022-07-13:
In the V2 push-wall, pick-place-wall, push-back, and shelf-place environments, setting `rand_init` to False and then resetting the environment would lead to the following error when calling `step()`:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/beegfs/home/zchuning/metaworld/metaworld/envs/mujoco/mujoco_env.py", line 25, in inner
    return func(*args, **kwargs)
  File "/mnt/beegfs/home/zchuning/metaworld/metaworld/envs/mujoco/sawyer_xyz/v2/sawyer_push_wall_v2.py", line 79, in evaluate_state
    ) = self.compute_reward(action, obs)
  File "/mnt/beegfs/home/zchuning/metaworld/metaworld/envs/mujoco/sawyer_xyz/v2/sawyer_push_wall_v2.py", line 157, in compute_reward
    object_grasped = self._gripper_caging_reward(
  File "/mnt/beegfs/home/zchuning/metaworld/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py", line 572, in _gripper_caging_reward
    caging_xz_margin = np.linalg.norm(self.obj_init_pos[xz] - self.init_tcp[xz])
TypeError: list indices must be integers or slices, not list
```
This is because the `reset()` functions of these environments set `self.obj_init_pos` to `self.adjust_initObjPos(self.init_config['obj_init_pos'])` when `self.random_init` is set to False (e.g. line 112 in `sawyer_push_wall_v2.py`). However, `self.adjust_initObjPos()` returns a python list instead of a numpy array. So line 572 of `swyer_xyz_env.py` triggers an error by accessing a list with a list of indices.

A simple fix is to wrap a `np.array()` around `self.adjust_initObjPos(self.init_config['obj_init_pos'])`, similar to how some other environments (e.g. Push V2) handle calls to `fix_extreme_obs_pos()`. But there might be a more systematic way to fix this, hence the github issue instead of a pull request.

krzentner, 2023-01-26:
We should probably delete the random_init flag, since it should always be set to True. It no longer does what it says it does anyways, since `_freeze_rand_vec` (which defaults to true) prevents goal randomization from happening on reset anyways.

krzentner, 2023-01-26:
Copied from Discord:
> Basically, there's a flag called `random_init` that only works when set to True, but does not actually cause initialization to be randomized.
> So it should be deleted and all of the code that uses it should be simplified using the assumption that it's always set to True.
> (Actual random initialization is controlled by the _freeze_rand_vec flag here: https://github.com/Farama-Foundation/Metaworld/blob/master/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py#L470 )

reginald-mclean, 2023-02-07:
Done!

krzentner, 2023-03-22:
The flag is still there, so I'm re-opening this issue.

reginald-mclean, 2023-03-22:
Currently we are trying to figure out the best way to maintain both the current V2 version of Meta-World while also having the ability to point users to the version of Meta-World that was used to create the Meta-World paper or other Meta-World results. For that reason, the changes for this issue have not been pushed to main. They can be found in the fixing-issues branch (which also contains some MuJoco updates).

closed 2023-03-22 by reginald-mclean