Farama-Foundation/Metaworld #377: TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

yaoxt3, 2022-07-28:
Hi, I try to import metaworld in python3.8.5, but the typeError issue occurs:
`>>> import metaworld
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/yaoxt3/RL/garage/metaworld/metaworld/__init__.py", line 7, in <module>
    import metaworld.envs.mujoco.env_dict as _env_dict
  File "/data/yaoxt3/RL/garage/metaworld/metaworld/envs/mujoco/env_dict.py", line 6, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.v1 import (
  File "/data/yaoxt3/RL/garage/metaworld/metaworld/envs/mujoco/sawyer_xyz/v1/__init__.py", line 1, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.v1.sawyer_assembly_peg import SawyerNutAssemblyEnv
  File "/data/yaoxt3/RL/garage/metaworld/metaworld/envs/mujoco/sawyer_xyz/v1/sawyer_assembly_peg.py", line 5, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.sawyer_xyz_env import SawyerXYZEnv, _assert_task_is_set
  File "/data/yaoxt3/RL/garage/metaworld/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py", line 11, in <module>
    from metaworld.envs.mujoco.mujoco_env import MujocoEnv, _assert_task_is_set
  File "/data/yaoxt3/RL/garage/metaworld/metaworld/envs/mujoco/mujoco_env.py", line 31, in <module>
    class MujocoEnv(gym.Env, abc.ABC):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases`

yaoxt3, 2022-07-28:
Does anyone else encounter this problem?

kanishkg, 2022-07-28:
OpenAI changed the Gym API recently, so using gym>=0.25.0 is probably breaking metaworld. Try downgrading gym to 0.24 and check?

yaoxt3, 2022-07-29:
Thank you very much! It works.

closed 2022-07-29 by yaoxt3