Farama-Foundation/Metaworld #379: Really High successful rate for RL2-PPO in ML45

waterhorse1, 2022-08-14:
Thanks for offering us this wonderful environments and benchmarks. When I checked the new benchmark results of ML10/ML45 on the updated paper, I found that the new RL2-PPO performance is really great and it reaches 70% successful ratio. It is kind of weird for me because RL2-PPO can achieve much great performance compared with all PPO based MT baselines in a large margin. 

As far as I know, the task id information is masked for RL2-PPO so it needs to interact with the environments for a few trajectories to infer the task information. It contains less information compared with MTPPO, but its performance is much better than MTPPO. Do you have any reasons or hypthesis for this phenmenon?

avnishn, 2022-08-15:
Yeah that's probably because there's a lot of information that a rnn like the one in rl2 captures dynamics between timesteps. 

We tried to overcome this by stacking consecutive timesteps in our mtppo experiments, but I suspect that more framestacking is required.

waterhorse1, 2022-08-15:
Thanks for your reply. Btw I want to double check that, in MT10/50 the initialisation and goal are fixed while in ML10/45 the initialisation and goal are random generated and the corresponding goal state are masked in the observation space. It this right?

waterhorse1, 2022-08-15:
It is also quite weird because, rl2-ppo is using ppo while mt-ppo is really poor, mt-sac is good but pearl is bad.... completely make no sense for me....

avnishn, 2022-08-16:
Yeah PEARL in general we found didn't scale well to complex environments and there are predecessors to the algorithm that were published. It was also very slow (the experiments when I ran them ran for v2 took 3 weeks to run on large single node gcp instances) 

I'm going to go ahead and close this for now since this isn't an issue with the metaworld.

closed 2022-08-16 by waterhorse1