Farama-Foundation/Metaworld #381: TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

victorgabillon, 2022-08-18:
Hello

Right after installation of the metaworld, when importing into python (for instance running 'import metaworld'), the following error appears:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/victor/metaw/lib/python3.9/site-packages/metaworld/__init__.py", line 7, in <module>
    import metaworld.envs.mujoco.env_dict as _env_dict
  File "/home/victor/metaw/lib/python3.9/site-packages/metaworld/envs/__init__.py", line 1, in <module>
    from metaworld.envs.mujoco.env_dict import (ALL_V2_ENVIRONMENTS_GOAL_HIDDEN,
  File "/home/victor/metaw/lib/python3.9/site-packages/metaworld/envs/mujoco/env_dict.py", line 6, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.v1 import (
  File "/home/victor/metaw/lib/python3.9/site-packages/metaworld/envs/mujoco/sawyer_xyz/v1/__init__.py", line 1, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.v1.sawyer_assembly_peg import SawyerNutAssemblyEnv
  File "/home/victor/metaw/lib/python3.9/site-packages/metaworld/envs/mujoco/sawyer_xyz/v1/sawyer_assembly_peg.py", line 5, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.sawyer_xyz_env import SawyerXYZEnv, _assert_task_is_set
  File "/home/victor/metaw/lib/python3.9/site-packages/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py", line 11, in <module>
    from metaworld.envs.mujoco.mujoco_env import MujocoEnv, _assert_task_is_set
  File "/home/victor/metaw/lib/python3.9/site-packages/metaworld/envs/mujoco/mujoco_env.py", line 31, in <module>
    class MujocoEnv(gym.Env, abc.ABC):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

using python 3.9

(note that a similar issue happened in the mtenv repository and the issue was solved somehow in the last commit)
for the discussion: https://github.com/facebookresearch/mtenv/issues/14
for the fix : https://github.com/facebookresearch/mtenv/compare/main...v2.0.0

Do you have an idea of what is happening?
Thanks for your help

YouJiacheng, 2022-08-22:
This should be fixed in upstream: https://github.com/openai/gym/pull/3001. You can upgrade your `gym`.

avnishn, 2022-08-24:
I'm hesitant to update gym versions right now because of the changes in the base gym api. If you'd like to submit a pr that would be great :)

YouJiacheng, 2022-08-24:
Actually there is no need to update gym macro or minor version, only need to skip two micro version. i.e. 0.25.0 and 0.25.1
and currently metaworld doesnt pin gym version, so just let user update their gym is enough.

avnishn, 2022-08-24:
Thanks for clarifying!

closed 2022-08-24 by avnishn