Farama-Foundation/Metaworld #399: Access to RGB images

tattrongvu, 2023-04-13:
Could someone provide a tutorial to take the RGB observation maybe from different positions from these environments, try to find it on the internet but not found a single one. Thanks very much, it will help a lot.

jren44, 2023-06-21:
Here's a snippet from how I was able to get the RGB image when evaluating on-policy (boilerplate code credit to the [CORL](https://github.com/tinkoff-ai/CORL/tree/main/algorithms) github). However, this approach is a bit buggy in the single-environment setting, as I frequently get segmentation faults when calling `env_cls()`. However, if I don't call `env_cls()`, then I get a "max timesteps exceeded" error. Somehow just `env.reset()` isn't cutting it. Hope this helps a little, and would be great if anyone has ideas about the aforementioned issues.

`@torch.no_grad()
def eval_actor(
    env_cls,
    actor: nn.Module,
    state_mean: None,
    state_std: None,
    use_img: bool,
    device: str,
    n_episodes: int,
) -> np.ndarray:
    actor.eval()
    num_success = 0
    episode_rewards = []
    seeds = np.arange(n_episodes)
    for i in range(n_episodes):
        env = env_cls()
        env.seed(int(seeds[i]))
        state, done = env.reset(), False
        episode_reward = 0.0
        for _ in range(env.max_path_length):
            if use_img:
                state = env.sim.render(camera_name="corner2", width=64, height=64)
                # state = np.transpose(state, (2, 0, 1)) / 255.0
                state = TEST_TIME_IMG_TRANSFORM(state).cpu().numpy()
            if state_mean is not None:
                state = (state - state_mean) / state_std
            action = actor.act(state, device)
            state, reward, done, info = env.step(action)
            episode_reward += reward
            if done or info["success"]:
                break
        if info["success"]:
            episode_rewards.append(episode_reward)
            num_success += 1
        else:
            episode_rewards.append(0)

    actor.train()
    return np.asarray(episode_rewards), num_success
`

reginald-mclean, 2023-06-21:
@tattrongvu you should examine the render function in [mujoco_env.py](https://github.com/Farama-Foundation/Metaworld/blob/b2a4cbb98e20081412cb4cc7ae3d4afc456a732a/metaworld/envs/mujoco/mujoco_env.py#LL125C4-L125C4)

reginald-mclean, 2023-06-21:
@jren44 I am not sure why you are getting a seg fault. What commit of Meta-World are you using? 
You are getting the error "max timesteps exceeded" because there is a maximum number of steps that you can take in the environment. If you call env.reset(), it should reset the step counter. There was a commit that didn't reset that counter, but it has been fixed in the most recent commit.

jren44, 2023-06-21:
Thanks for the reply @reginald-mclean ! The version I am currently using is: 
`metaworld @ git+https://github.com/Farama-Foundation/Metaworld.git@3ced29c8cee6445386eba32e92870d664ad5e6e3`, which I just installed using `pip install --upgrade git+https://github.com/Farama-Foundation/Metaworld.git@master#egg=metaworld` today. However, I'm still getting the `ValueError: Maximum path length allowed by the benchmark has been exceeded`.

jren44, 2023-06-21:
Thanks for the heads up @reginald-mclean . I pulled the latest commits, but for some reason am still getting a seg fault on the line `state = env.sim.render(camera_name="corner2", width=64, height=64)`. Curious if I'm using the render function incorrectly?

reginald-mclean, 2023-06-21:
You should just be able to do env.render(camera_name="corner2", width=64, height=64)

reginald-mclean, 2023-06-27:
@jren44 did you resolve this?

jren44, 2023-06-28:
Yes! We talked it out on Discord 😄. Thanks!

reginald-mclean, 2023-06-28:
@jren44 didn't realize that was you! No problem

closed 2023-06-28 by reginald-mclean