Farama-Foundation/Metaworld #403: About the specific meaning of each parameter in the state？

wyqsss, 2023-05-30:
The state is a vector of length 39, but what does each parameter represent specifically? Where can I find this information?

pseudo-rnd-thoughts, 2023-05-30:
What environment in particular are you talking about?

wyqsss, 2023-05-30:
> What environment in particular are you talking about?

Is there a significant difference in the meaning of parameters in different environments? For example, I want to know the meaning of the parameters for Door-open and Window-open. Is there a document that describes the meaning of state parameters in all environments?

pseudo-rnd-thoughts, 2023-05-30:
@reginald-mclean Will know the answer but is on holiday until the end of the week I believe

wyqsss, 2023-05-30:
> @reginald-mclean Will know the answer but is on holiday until the end of the week I believe

Okay, thank you for your response.

reginald-mclean, 2023-06-01:
The state is a vector of size 36 with the first 18 elements being the current observation and the next 18 elements being the previous observation. The first observation after doing an env.reset() will have the same vector for both of the 18 elements.

Each group of 18 elements can be broken down into the following items: gripper xyz position, single element that describes how open/closed the gripper is, object 1 xyz position, object 1 quaternion, object 2 xyz position, and object 2 quaternion. If a field isn't used, then it is zeroed out (ie only one object, then the object 2 fields are zeroes). In summary [(gripper x, y, z), (open/close), (obj1 x, y, z), (obj 1 quaternion a, b, c, d), (obj 2 x, y, z), (obj2 quaternion a, b, c, d)]. 

If you are doing multi-task learning than this is a vector of size 39 with the addition of the goal as the last 3 elements of the vector.

wyqsss, 2023-06-01:
> The state is a vector of size 36 with the first 18 elements being the current observation and the next 18 elements being the previous observation. The first observation after doing an env.reset() [...]

Thank you very much for your response.We also want to know the meaning of each parameter in the action.

reginald-mclean, 2023-06-01:
If you look at the [step function](https://github.com/Farama-Foundation/Metaworld/blob/e5804e300db0ca4235aeecfd7557c93abfec8622/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py#LL409C28-L409C28) the first three elements is used to calculate the position delta, and then the last element is used as a control force sent to the actuators in the simulation.

closed 2023-06-01 by reginald-mclean