Farama-Foundation/Metaworld #404: Adopt Semantic Versioning and create new releases

lebrice, 2023-05-30:
Hey there! Seems like recent commits might changed some of the behaviour of some environments on `reset` (Another issue will be made about this soon).

Would it be possible for this project to adopt a semantic versioning scheme, and to publish new releases at some point?

Thanks a lot!

pseudo-rnd-thoughts, 2023-05-30:
The issue is that `setup.py` has `gym >= 0.15` and `gym` has had a number of API changes since `0.15`. I would suggest reverting back to gym < `0.21`

closed 2023-07-11 by reginald-mclean

lebrice, 2023-07-11:
> The issue is that `setup.py` has `gym >= 0.15` and `gym` has had a number of API changes since `0.15`. I would suggest reverting back to gym < `0.21`

Wouldnt it make more sense to instead change setup.py to restrict it to the valid gym versions then?