Farama-Foundation/Metaworld #405: A recent commit introduced a bug in Metaworld v2 - env.step() doesn't return done when curr_path_length == max_path_length

optimass, 2023-05-30:
Hi,

I can't exactly pinpoint where the bug was introduced but, I recently pulled a new version of Metaworld and a bug was introduced.
Specifically, while running an episode, env.step() would never return Done=True, thus the code would throw an error at step 501 (max_path_length = 500 by default).
It might have to do w/ a recent commit, e.g. [here ](https://github.com/Farama-Foundation/Metaworld/commit/82cecfb24eed51c0c4a79a5e06fee1cd36892b15#r115559039) you can see that self.curr_path_length is never reset when env.reset() is called. Now, that's not exactly the source of the issue but maybe the bug was introduced in the same commit. 
I'm guessing there is a bug in env.step() w.r.t. the logic that checks self.curr_path_length == self.max_path_length

optimass, 2023-05-30:
here's the part of the error that's relevant:

```
File "/Users/massimo/opt/anaconda3/envs/tacrl/lib/python3.8/site-packages/metaworld/envs/mujoco/mujoco_env.py", line 25, in inner
    return func(*args, **kwargs)
  File "/Users/massimo/opt/anaconda3/envs/tacrl/lib/python3.8/site-packages/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py", line 411, in step
    self.do_simulation([action[-1], -action[-1]])
  File "/Users/massimo/opt/anaconda3/envs/tacrl/lib/python3.8/site-packages/metaworld/envs/mujoco/mujoco_env.py", line 110, in do_simulation
    raise ValueError('Maximum path length allowed by the benchmark has been exceeded')
ValueError: Maximum path length allowed by the benchmark has been exceeded
```

reginald-mclean, 2023-06-01:
Hi @optimass. Meta-World has never returned Done = True when the task has been completed. I don't know the exact reasoning behind that as I am not the original maintainer and can't find any great documentation of this. You can monitor the info['success'] field for completion. When info['success'] == 1.0 then the task is complete, 0.0 otherwise.

reginald-mclean, 2023-06-01:
The Maximum path length issue was introduced with PR #401

closed 2023-06-01 by reginald-mclean