Farama-Foundation/Metaworld #409: Roadmap

reginald-mclean, 2023-06-06:
This is a roadmap of the current plans for Meta-World. Further information can be found here [Monday Board](https://farama.monday.com/boards/3855618447). The Meta-World project is undergoing updates for Quality of Life, new package support, and ease of use improvements. 

January-May

- [x] Upgrade to Deepmind MuJoco from mujoco_py

June
- [ ] Update Meta-World to Gymnasium from Gym
- [ ] Update for Python 3.8-3.11 support (dropping 3.7)
- [ ] Move Meta-World into Gymnasium-Robotics
- [ ] Create an Installation Guide

July
- [ ] Create a Rendering Guide
- [ ] Release v0.3.0

pseudo-rnd-thoughts, 2023-06-08:
Do we want to make a github release for the legacy metaworld project?

reginald-mclean, 2023-06-08:
@pseudo-rnd-thoughts I think that makes sense. I have no idea how to do that so I would need help.

pseudo-rnd-thoughts, 2023-06-08:
We have done it for miniwob++ project.
I believe you can make a release from a particular commit or branch.

If we know the point at which a legacy release makes sense, I.e. before farama took over then we should be able to do this

closed 2024-04-29 by reginald-mclean