Farama-Foundation/Metaworld #432: install version0.0.0

hjh0119, 2023-07-17:
I'm realizing the experiment of the paper "Transformers are Meta-Reinforcement Learners," and the required version of metaworld is v0.0.0. Could you please guide me on how to download it?

reginald-mclean, 2023-07-17:
they install it through pip in the repository: https://github.com/luckeciano/transformers-metarl/blob/5528cdb4ef88ff4021ce8386f0b0a1402530d9df/environment.yml#L75

hjh0119, 2023-07-19:
> they install it through pip in the repository: https://github.com/luckeciano/transformers-metarl/blob/5528cdb4ef88ff4021ce8386f0b0a1402530d9df/environment.yml#L75

The required version 0.0.0 of metaworld is not available in the pip repository.

```
ERROR: Could not find a version that satisfies the requirement metaworld==0.0.0 (from versions: 0.0.1.dev0)
ERROR: No matching distribution found for metaworld==0.0.0
```

reginald-mclean, 2023-07-19:
It doesn't look like it's on pypi as I thought, not sure why they pinned
that version. Our v2.0.0 release that's on GitHub is probably your best
bet. Old versions (pre the v2.0.0 version) are tough to get running

Reginald McLean (he/him)
Ph.D. Student, Department of Computer Science
Toronto Metropolitan University (formerly Ryerson University)
***@***.***
https://www.torontomu.ca/

hjh0119, 2023-07-20:
> It doesn't look like it's on pypi as I thought, not sure why they pinned that version. Our v2.0.0 release that's on GitHub is probably your best bet. Old versions (pre the v2.0.0 version) are tough [...]

I understand. Thank you for your patient response

closed 2023-07-20 by hjh0119