Farama-Foundation/Metaworld #436: Truncating when using new gymnasium version

goldbird5, 2023-07-30:
I'm new to meta-world, and what I understood is that, for version 2, there is no returning 'done' when neither it successes nor reaches max_path_length(which is currently 500).
I understand done is not needed for success because it would make agent to accumulate reward to the end.
However, truncation, which means terminating at 'self.curr_path_length == self.max_path_length' seems to be still needed for RL algorithms, so I should explicitly handle the truncation in my code(If it is intended to, I guess).

What if we handle truncation in meta-world code since recent version of gymnasium uses 5 return values for step function as:
'obs, reward, terminal, truncate, info = env.step()'

Also, as I mentioned, I'm new to metaworld, so I'm confused why the max_path_length increased from 150/200 to 500 now. I mean, should I use 500 for benchmark now?

In addition, there is a bug with rendering when accessing single goal environments as README.md. i.e.

```
from metaworld.envs import (ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE,
                            ALL_V2_ENVIRONMENTS_GOAL_HIDDEN)
                            # these are ordered dicts where the key : value
                            # is env_name : env_constructor

import numpy as np

door_open_goal_observable_cls = ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE["door-open-v2-goal-observable"]
door_open_goal_hidden_cls = ALL_V2_ENVIRONMENTS_GOAL_HIDDEN["door-open-v2-goal-hidden"]

env = door_open_goal_hidden_cls()
```

current gym env uses render_mode, but [initailize function](https://github.com/Farama-Foundation/Metaworld/blob/844d76da3a49b23bd1b78c0b9e4744e73af43495/metaworld/envs/mujoco/env_dict.py#L404) doesn't support that.

Thank you.

goldbird5, 2023-08-03:
@reginald-mclean Thanks for reply.

Yes, I tried to do so, but 'door_open_goal_hidden_cls(render_mode="human")' returns Error because 'initialize' function can't handle 'render_mode' in [here](https://github.com/Farama-Foundation/Metaworld/blob/844d76da3a49b23bd1b78c0b9e4744e73af43495/metaworld/envs/mujoco/env_dict.py#L404).

jity16, 2023-08-31:
A possible solution - change the create_observable_goal_envs() functions:

~~~
def initialize(env, render_mode=None, seed=None):
            if seed is not None:
                st0 = np.random.get_state()
                np.random.seed(seed)
            super(type(env), env).__init__()
            env.render_mode = render_mode
            env._partially_observable = False
            env._freeze_rand_vec = False
            env._set_task_called = True
            env.reset()
            env._freeze_rand_vec = True
            if seed is not None:
                env.seed(seed)
                np.random.set_state(st0)
~~~

And then you can make the env with the parameter render_mode=xxx.

reginald-mclean, 2023-10-04:
@jity16 @goldbird5 I just pushed an update for the render mode fix.