Farama-Foundation/Metaworld #437: Although seed is fixed, the randomness is still out of control.

seungju-k1m, 2023-08-05:
Hello.

Thanks for your wonderful project.

I found out that the randomness is out of control even if the seed is provided.

Here is snap-shot.

test_seed.py
```
import metaworld

seed = 777
task_id = "pick-place-v2"
task_idx = 0

benchmark = metaworld.ML1(task_id)
task = benchmark.train_tasks[task_idx]

env = benchmark.train_classes[task_id]()
env.set_task(task)
obs, _ = env.reset(seed=seed)
print(obs.mean())
```

Result
[image]

pseudo-rnd-thoughts, 2023-08-05:
I suspect that the task generated is not the same, could you print that? 
@reginald-mclean Is there anyway that we could add a seed value when generating a task? This is probably a question when improving the API

reginald-mclean, 2023-08-05:
If you're using an environment within one of the benchmarks (ie MT10, or ML1) then you need to seed the creation of the benchmark not the individual environment. When you create the benchmark use seed, ie env = benchmark.train_classes[task_id](seed=seed)

closed 2023-08-05 by pseudo-rnd-thoughts