Farama-Foundation/Metaworld #441: No "self.sim" for v2 envs

YanjieZe, 2023-10-04:
I need to use sim to do the rendering things, but I found that for the latest version, `sim` is not there... Could anyone explain? 
I check other branches and find that in some branches `sim` is still there. I am just not sure why such changes are made and what should I do to get the rendering in this condition.
```
ipdb> env.sim
*** AttributeError: 'SawyerNutAssemblyEnvV2' object has no attribute 'sim'
```

YanjieZe, 2023-10-04:
I switch to an earlier commit temporarily, but maybe this issue still should be noted.

reginald-mclean, 2023-10-04:
It's because of a change in the Mujoco bindings. Instead of using env.sim, when you create an environment you can pass the render_type argument to the constructor. I am in the process of updating the single environment creation to handle that argument as well.

closed 2023-10-04 by reginald-mclean