Farama-Foundation/Metaworld #448: Rendering with render_mode="rgb_array" does not work properly when rendering multiple envs

ManfredStoiber, 2023-10-11:
When using multiple environments, a strange behavior occurs during rendering in mode "rgb_array":
1. When calling the .render() function on one environment, it is displayed correctly
2. When calling it on a second environment, it also works as intended
3. When rendering the first env again, the objects have weird shapes (see output below)

Here is a minimum example for reproduction:

```
import metaworld
import random
import matplotlib.pyplot as plt

mt50 = metaworld.MT50() # Construct the benchmark, sampling tasks

training_envs = []
for name, env_cls in mt50.train_classes.items():
  env = env_cls(render_mode="rgb_array")
  task = random.choice([task for task in mt50.train_tasks
                        if task.env_name == name])
  env.set_task(task)
  env.camera_name="corner3"
  training_envs.append(env)

# 1. works (image 1)
plt.imshow(training_envs[34].render()) # pick-place-v2
plt.show()

# 2. works (image 2)
plt.imshow(training_envs[1].render()) # basketball-v2
plt.show()

# 3. does not work anymore (image 3)
plt.imshow(training_envs[34].render()) # pick-place-v2
plt.show()
```
Output:
[image]
[image]
[image]

Also the images are upside down when using different cameras than default, but that's also the case when using only one environment.

I noticed that the problem does not occur with every combination of environments, for example "pick-place-v2" and "reach-v2" work together, whereas "pick-place-v2" and "basketball-v2" don't. Maybe because the former combination use the same objects and the latter use different objects?

Thank you very much!

reginald-mclean, 2023-10-11:
Hi! Thanks for opening this issue. This is really interesting. It probably has something to do with Mujoco, not Meta-World itself. Let me investigate a little bit. I can reproduce the issue on my end. If it's not a Meta-World/Gymnasium issue you may have to open an issue with Mujoco.

reginald-mclean, 2023-10-11:
@ManfredStoiber this issue is because of the OpenGL context that is used to render by Mujoco. When you render in env1 the viewer is created and given the "current context." Rendering in env2 is the same. However when you go back to env1, the viewer is reused from the first render call and isn't explicitly given the current context back. If you look at the issue I opened on the [Gymnasium repo](https://github.com/Farama-Foundation/Gymnasium/issues/736) you can find a suggested fix that you can try. It seems to work on my machine but let me know if it doesn't work for you.

reginald-mclean, 2023-10-11:
Farama-Foundation/Gymnasium#736

ManfredStoiber, 2023-10-11:
It works! Thank you very much for your quick reply and your help.
Best regards

reginald-mclean, 2023-11-20:
@ManfredStoiber it seems like this is going to be fixed within Meta-World so I am going to re-open the issue so we can track it.

zhihou7, 2024-08-27:
Why the rendered image look like wired? fliped?

![](https://private-user-images.githubusercontent.com/47210077/274156400-732d7e25-3889-4be7-bcb4-af00d56bc2f9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NzQ3NzIsIm5iZiI6MTcyNDc3NDQ3MiwicGF0aCI6Ii80NzIxMDA3Ny8yNzQxNTY0MDAtNzMyZDdlMjUtMzg4OS00YmU3LWJjYjQtYWYwMGQ1NmJjMmY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDE2MDExMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlmY2ZlOTI4ZDZhNTFiZTA5YzE4OTkwNjhkMDk5ODAxZDRhZDk2NTgzZjVlYzZjNmM1Nzg4ZTJhODAxMDM4MDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.D75OhhJ0N1dwB1kNUAOPrTZZEEbK0hXCluFCnb-yaCY)

Kallinteris-Andreas, 2024-10-24:
Is this bug still present with gymnasium 1.0 and the latest version of meta world?

AvalonGuo, 2025-09-27:
> [@ManfredStoiber](https://github.com/ManfredStoiber) it seems like this is going to be fixed within Meta-World so I am going to re-open the issue so we can track it.

Hi @ManfredStoiber 👋

I'm also hitting this issue 🙃

Any update on the fix in Meta-World?

Here's a minimal repro:
`import gymnasium as gym
import numpy as np
import metaworld
import matplotlib.pyplot as plt
from metaworld.env_dict import MT10_V3
from metaworld.evaluation import _get_task_names
from msdc.environment import get_task_metadata,get_task_descriptions

train_envs = gym.make_vec("Meta-World/MT10",max_episode_steps=500,vector_strategy='sync',seed=0,render_mode="rgb_array",camera_name="corner2")
# task_metadata = get_task_metadata(train_envs)
task_names = _get_task_names(train_envs)
obs,info = train_envs.reset()
images = train_envs.render()
for i, image in enumerate(images):
    image = image[::-1,:,:]
    plt.figure(figsize=(8, 8))
    plt.imshow(image)
    plt.title(f"Environment {task_names[i]}")
    plt.axis('off')
    plt.show()

actions = train_envs.action_space.sample()
train_envs.step(actions)
images = train_envs.render()
for i, image in enumerate(images):
    image = image[::-1,:,:]
    plt.figure(figsize=(8, 8))
    plt.imshow(image)
    plt.title(f"Environment {task_names[i]}")
    plt.axis('off')
    plt.show()`