Farama-Foundation/Metaworld #455: Camera view?

Hong753, 2023-12-07:
Does anyone know how to change the camera view when rendering? 
It seems there are several camera view modes (topview, corner, etc.), but i can't find a way to set the camera_name attribute, or make it work.

reginald-mclean, 2024-01-12:
Hi @Hong753. I think there is an issue passing the selected camera from MW -> Gymnasium. I had to modify the Gymnasium mujoco_env.py render function to accept arguments then I was able to render from the desired camera.

closed 2024-01-12 by reginald-mclean

varungupta31, 2024-08-05:
@reginald-mclean hey, can you please tell more about the change you mentioned?
I tried to follow the code trail, but couldn't quite figure it out.

I found camera_id to be passed in some places, and though that would work, but that too didn't change the camera view.

Thanks,