Farama-Foundation/Metaworld #460: ValueError: could not broadcast input array from shape (11,) into shape (7,)

nanzhu2003, 2024-01-23:
When I run the ML1 code in the example, I get the following error:
[image]
My package is installed already and my gmnasium version is 0.29.1 and metaworld is 2.0.0, the system I'm using is ubuntu 20.04.I don't understand why I'm having this problem. Thank you very much for your help!

reginald-mclean, 2024-01-23:
Hi! This seems like a mismatch between Mujoco versions. With Mujoco-Py the mocap welds was an array of size 7, in Mujoco the mocap welds is an array of size 11. I would check that first. If that doesn't fix it, can you provide a script to re-produce and the output of `pip list` ?

nanzhu2003, 2024-01-23:
> Hi! This seems like a mismatch between Mujoco versions. With Mujoco-Py the mocap welds was an array of size 7, in Mujoco the mocap welds is an array of size 11. I would check that first. If that doesn't fix it, can you provide a script to re-produce and the output of `pip list` ?

Of course, my mujoco is 2.2.0 and my mujoco_py is 2.1.2.14, and one of the blogs I found taught me to install it that way, so is there a problem with that? Thank you very much!
[image]

nanzhu2003, 2024-01-23:
If needed the full output of my pip list is as follows：
[image]
[image]
The error arises when I run this code:
import metaworld
import random

print(metaworld.ML1.ENV_NAMES)  # Check out the available environments

ml1 = metaworld.ML1('pick-place-v2') # Construct the benchmark, sampling tasks

env = ml1.train_classes['pick-place-v2']()  # Create an environment with task `pick_place`
task = random.choice(ml1.train_tasks)
env.set_task(task)  # Set task

obs = env.reset()  # Reset environment
a = env.action_space.sample()  # Sample an action
obs, reward, done, info = env.step(a)  # Step the environment with the sampled random action

nanzhu2003, 2024-01-23:
> Hi! This seems like a mismatch between Mujoco versions. With Mujoco-Py the mocap welds was an array of size 7, in Mujoco the mocap welds is an array of size 11. I would check that first. If that doesn't fix it, can you provide a script to re-produce and the output of `pip list` ?

I found something about it saying that mujoco_py doesn't support mujoco versions after 2.1.0, which I think would be the cause of the problem. Thank you very much！

nanzhu2003, 2024-01-23:
Hi, the error arises when I run this code:

import metaworld
import random

print(metaworld.ML1.ENV_NAMES)  # Check out the available environments

ml1 = metaworld.ML1('pick-place-v2') # Construct the benchmark, sampling tasks

env = ml1.train_classes['pick-place-v2']()  # Create an environment with task `pick_place`
task = random.choice(ml1.train_tasks)
env.set_task(task)  # Set task

obs = env.reset()  # Reset environment
a = env.action_space.sample()  # Sample an action
obs, reward, done, info = env.step(a)  # Step the environment with the sampled random action

reginald-mclean, 2024-01-24:
You don't need both mujoco-py and mujoco. If you download the Meta-World repo you should be able to do `pip install -e .' from inside the MW folder. It should install everything you need to get the code running

reginald-mclean, 2024-01-24:
Are you using a version of Meta-World that isn't the most recent commit?

nanzhu2003, 2024-01-25:
> You don't need both mujoco-py and mujoco. If you download the Meta-World repo you should be able to do `pip install -e .' from inside the MW folder. It should install everything you need to get the code running

I did this according to the documentation and downloaded the latest version but the problem exists, I tried uninstalling mujoco-py and it still exists. It's really hard to fix.

nanzhu2003, 2024-01-25:
I uninstalled my mujoco_py and tried `pip install -e' and found the problem solved. I was using install setup.py before. thank you very much for your help! Have a nice day!

nanzhu2003, 2024-01-25:
> I reinstalled Metaworld as you did in one of my new conda environments, and the above problem is solved, but with:
> 
> ```
> obs, reward, done, info = env.step(a) # Step the environment with the sampled random action
> ValueError: too many values to unpack (expected 4)
> ```
> 
>Finally, I compare the versions of mujoco between my different conda environment, and find that if I update the mujoco version from 2.1.4 to 2.3.7, the problem as the title solved. However， I don't know how to solve the problem 'too many value to unpack'. I'm new to RL and it really disturb me. In any case, I sincerely thank you for your great help!

reginald-mclean, 2024-01-25:
This is because the return from env.step(a) has been updated to:
`obs, reward, truncation, termination, info = env.step(a)` but we haven't updated those examples!

closed 2024-01-25 by reginald-mclean

nanzhu2003, 2024-01-26:
> This is because the return from env.step(a) has been updated to: `obs, reward, truncation, termination, info = env.step(a)` but we haven't updated those examples!

OK, sincerely thank you!