Farama-Foundation/Metaworld #461: `SawyerXYZEnv._get_state_rand_vec` is not seeded

dtch1997, 2024-01-28:
`SawyerXYZEnv` has a method `.seed()`. Intuitively, this is supposed to ensure that seeded env creation is deterministic.

However, this is not the case at the moment. I wrote this simple failing test:

```python
from metaworld.envs import ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE

MetaworldEnvName = str

def setup_metaworld_env(
    env_name: MetaworldEnvName, 
    seed: int = 0,
) -> GymEnv:
    env_cls = ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE[env_name]
    e: SawyerXYZEnv = env_cls(render_mode="rgb_array")

    # Hack: enable random reset
    e._freeze_rand_vec = False
    e.seed(seed)
    return e

def test_seeded_env_creation_is_deterministic():
    env1 = setup_metaworld_env("drawer-open-v2-goal-observable", "top_cap2", seed=0)
    obs1, _ = env1.reset()
    img1 = env1.render()
    
    env2 = setup_metaworld_env("drawer-open-v2-goal-observable", "top_cap2", seed=0)
    obs2, _ = env2.reset()
    img2 = env2.render()

    assert np.allclose(obs1, obs2)
    assert np.allclose(img1, img2)
```

dtch1997, 2024-01-28:
I traced the error to this section: https://github.com/Farama-Foundation/Metaworld/blob/c822f28f582ba1ad49eb5dcf61016566f28003ba/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py#L547C1-L551C33

We should use `self.np_random` generator, which is seeded by `.seed()`, as opposed to using the unseeded `np.random` generator. After making this change the test passes.

reginald-mclean, 2024-01-30:
@dtch1997 thank you for pointing that out. If you want to make a PR to fix this, feel free

closed 2024-04-03 by reginald-mclean