Farama-Foundation/Metaworld #472: About camera size of observation

yingchengyang, 2024-04-19:
Hi~ I'm curious about the image size of the observation. When I create an env like below

```
mt1 = MT1(task_name, seed=seed) 
env = mt1.train_classes[task_name](render_mode='rgb_array')
env.set_task(mt1.train_tasks[task_id]) 
```

How can I change the img size of the observation in the env? Thanks a lot!

reginald-mclean, 2024-04-19:
Hi! I haven't tried this so I am not 100% sure it will work, but you should be able to pass a desired height and width into the mujoco renderer. If you go to [this line](https://github.com/Farama-Foundation/Metaworld/blob/0745f6c905f7823747a8e6b9fb5580909bb5397c/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py#L37) that init function calls the gymnasium mujoco env. If you add height=X, width=Y, you should be able to manipulate the image size.

yingchengyang, 2024-04-19:
Thanks a lot! How can I go into this function to change the size? Are there some external parameters that can be controlled? By the way, I find that in some codebases, we can use the following code to create an env with given img size:

```
mw_gym_make(name, cam_height=img_size, cam_width=img_size, cam_name=cam_name, train_distrib=train, seed=seed)
```

Unfortunately, I find that the function mw_gym_make is deleted in the lasted version of metaworld. Are there any similar functions? Thanks a lot!

reginald-mclean, 2024-04-19:
No there aren't any similar functions in MW right now

closed 2024-04-23 by reginald-mclean