Farama-Foundation/Metaworld #477: copy.deepcopy fails to correctly copy an environment

Zhao3miao, 2024-05-03:
When using the following code:
```
import copy
import metaworld
import random

ml1 = metaworld.ML1('pick-place-v2')
env = ml1.train_classes['pick-place-v2']()
task = random.choice(ml1.train_tasks)
env.set_task(task)
env2 = copy.deepcopy(env)
print(env.action_space)
print(env2.action_space)
```

The console outputs:

```
Box(-1.0, 1.0, (4,), float32)
Box(-1.0, 1.0, (2,), float32)
```

It seems that the copy.deepcopy function is not correctly copying the environment. The action spaces of the original environment and the copied environment are different.

Environment Configuration:
> Python 3.8
> Mujoco 3.1.4
> Mujoco-py 2.1.2.14
> Ubuntu 20.04.4
> Gymnasium 0.29.1

Zhao3miao, 2024-05-03:
In [Garage](https://github.com/rlworkgroup/garage), both LocalSampler and RaySampler cannot be used due to this issue.

reginald-mclean, 2024-05-03:
I would suggest trying release 2.0.0 as well, but if I remember correctly this issue exists with 2.0.0 as well.

Zhao3miao, 2024-05-04:
> I would suggest trying release 2.0.0 as well, but if I remember correctly this issue exists with 2.0.0 as well.

yes, mujoco 2.0.0 have this issue too

Zhao3miao, 2024-05-04:
Thank for your reply and help! Release 2.0.0 works well with copy.deepcopy
best wishes

closed 2024-05-04 by Zhao3miao

zpw130, 2025-03-17:
Based on my experimentation, release 2.0.0 still exhibits this issue. After multiple attempts, I discovered the root cause: when using deepcopy to replicate environments, it triggers the self._set_action_space() function from the parent class MujocoEnv of SawyerXYZEnv, which incorrectly changes the action_space to Box(-1.0, 1.0, (2,), float32). A viable solution is to override the self._set_action_space() method in the SawyerXYZEnv class. The specific implementation steps are as follows:
1. Add the following code to the SawyerXYZEnv class in site-packages/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py:
`def _set_action_space(self):
    self.action_space = Box(  # type: ignore
        np.array([-1, -1, -1, -1]),
        np.array([+1, +1, +1, +1]),
        dtype=np.float32,
    )`
2. In the init section of the SawyerXYZEnv class, replace the original code block:
`self.action_space = Box(  # type: ignore
    np.array([-1, -1, -1, -1]),
    np.array([+1, +1, +1, +1]),
    dtype=np.float32,
)`
with:
`self._set_action_space()`
This modification ensures the action_space is correctly maintained when using deepcopy replication.