Farama-Foundation/Metaworld #479: TypeError: render() got an unexpected keyword argument 'offscreen'

Taskii-Lei, 2024-05-14:
Thanks for your great work! I love the Metaworld!
I want to render the env, and resize it to (64,64). But when I run ```env.render(offscreen=True, resolution=(64,64), camera_name="corner")```, it raises error: ```TypeError: render() got an unexpected keyword argument 'offscreen'```. And so do the parameter 'resolution' . I have tried [https://github.com/Farama-Foundation/Metaworld/issues/34](#34), but it doesn't work for me.

The Virtual Env:
I followed the manual configuration, ran ``` pip install -e . ```  and succeeded.  And I have ```apt install xvfb```

Machine:
I run it on a remote machine.

Codes:
```
import metaworld
benchmark = metaworld.ML1("pick-place-v2")
env = benchmark.train_classes["pick-place-v2"]()
image = env.render(offscreen=True, resolution=(64,64), camera_name="corner")
print(image.shape)
```

ERROR:
File "/xxx/x/xxxx/test.py", line 4,  in <module>
    env.render(offscreen=True, resolution=(64,64), camera_name="corner").shape
TypeError: MujocoEnv.render() got an unexpected keyword argument 'offscreen'

I was so confused if it's because any lib is missing or not installed? or versions mismatch?

BTW, the ```env.step(a)``` is not correct, cuz. I found the output is a 5-tuple, not 4-tuple. I read the source code  and found it should be (see Metaworld/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py:614~620):
```
obs, reward, False, truncate, info = env.step(a)
```
The  ```truncate``` is set as a done-flag, and the third one, which should be done-flag is set always False. The [https://github.com/Farama-Foundation/Metaworld?tab=readme-ov-file#using-the-benchmark](UsingTheBenchmark) in README.md should be corrected or at least should be illustrated because it's a little bit misleading. Thank you !!

reginald-mclean, 2024-05-14:
Hi @Taskii-Lei. Thank you for opening this issue. You are correct that the README.md needs to be updated :) 

The API has changed slightly, it should now be something like:

`
import metaworld
benchmark = metaworld.ML1("pick-place-v2")
env = benchmark.train_classes["pick-place-v2"](camera_name='corner')
image = env.render()
print(image.shape)
`
We don't get pass any height or width fields to the Gymnasium MujocoEnv class as of right now, so you will need to modify the [Sawyer_XYZ base class](https://github.com/Farama-Foundation/Metaworld/blob/83ac03ca3207c0060112bfc101393ca794ebf1bd/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py#L50) as well:
`
mjenv_gym.__init__(
            self,
            model_name,
            frame_skip=frame_skip,
            observation_space=self.sawyer_observation_space,
            render_mode=render_mode,
            camera_name=camera_name,
            camera_id=camera_id,
            height = X,
            width = X,
        )
`

where X and Y are your desired render sizes. Let me know if that doesn't work!

Taskii-Lei, 2024-05-15:
I modified some parameters of related classes according to your tips, and now it runs very well! Thank you so much!

reginald-mclean, 2024-05-15:
I'm going to close this issue, but feel free to open it if you find any other issues!

nicklashansen, 2025-06-12:
Hi @reginald-mclean, I'm updating my Meta-World code from gym to gymnasium and am running into this issue as well. Would it be possible to add support for desired render resolution again? It is inconvenient having to package custom env codebases just for this functionality :-)

nicklashansen, 2025-06-13:
Yes I think that should be sufficient, thank you! I'll give it a try soon and report back. Being able to specify resolution is important for image-based RL.

closed 2025-06-18 by reginald-mclean