Farama-Foundation/Metaworld #487: env.render() returns None

jayaramreddy10, 2024-07-22:
Hi, 

I am trying to render image during rollout of my SAC policy using command:  frame = env.render(mode='rgb_array')

And frame turns out to be None rather than image.

But when I use the command : self._get_viewer('human').render()  , I am able to see the rendering on the screen but unable to save it as image.

How do I get the rendered image? Do I need to add any other additional arguments in this command:  frame = env.render(mode='rgb_array') ?

reginald-mclean, 2024-07-22:
Hi sorry, our documentation needs to be updated. Follow the examples here: https://metaworld.farama.org/rendering/rendering/
You just need to set a render mode upon creation of the environment and then call
`frame = env.render(mode='rgb_array')`

Kallinteris-Andreas, 2024-07-22:
Render mode should be set during environment initialization. Check the documentation for examples

jayaramreddy10, 2024-07-23:
Thanks , its working now

closed 2024-07-23 by pseudo-rnd-thoughts