Farama-Foundation/Metaworld #490: Issues with Rendering - no attribute 'glfwGetCurrentContext' [MacOS]

varungupta31, 2024-08-05:
I am new to this benchmark and tool.

I installed as per the documentation

```
pip install git+https://github.com/Farama-Foundation/Metaworld.git@master#egg=metaworld
```

And tried the started code,

```python
import metaworld
import random

print(metaworld.ML1.ENV_NAMES)  # Check out the available environments

env_name = 'basketball-v2' # Pick an environment name

render_mode = 'human' # set a render mode

ml1 = metaworld.ML1(env_name) # Construct the benchmark, sampling tasks

env = ml1.train_classes[env_name](render_mode=render_mode)
task = random.choice(ml1.train_tasks)
env.set_task(task)  # Set task

obs = env.reset()  # Reset environment
a = env.action_space.sample()  # Sample an action
obs, reward, terminate, truncate, info = env.step(a)
```

I run into this error,
```
Exception ignored in: <function WindowViewer.__del_ at 0x10724c310>
Traceback (most recent call last):
File "/opt/homebrew/Caskroom/miniconda/base/envs/meta/lib/python3.10/site-pack ages/gymnasium/envs/mujoco/mujoco_rendering.py", line 359, in _del_
File "/opt/homebrew/Caskroom/miniconda/base/envs/meta/lib/python3.10/site-pack
ages/gymnasium/envs/mujoco/mujoco_rendering.py", line 352, in free
File "/opt/homebrew/Caskroom/miniconda/base/envs/meta/lib/python3.10/site-pack
ages/glfw/_
-
init
_py", line 2366, in get_current_context
AttributeError: 'NoneType' object has no attribute 'glfwGetCurrentContext'
```

A similar error was observed on Ubuntu 20.04, with the same setup instruction and code.

Kindly help, 
Thanks.

reginald-mclean, 2024-08-06:
That's not an error that causes problems, it happens when your script ends without explicitly calling `env.close()`

varungupta31, 2024-08-08:
Thanks, figured that out ... rookie mistake 🫠