Farama-Foundation/Metaworld #496: Non deterministic behaviour of env.reset

jayaramreddy10, 2024-08-26:
Hi, 

I am not able to see the same state after reset. I am trying to run the episode multiple times with same checkpoint but initial state is changing after call to env.reset() at the beginning of episode even after fixing the seed. Could you please suggest a solution for this?

Code to replicate the issue: 

    seed = cfg.seed
    set_seed(seed)

    # env=build_environment(cfg)
    checkpoint_dir = 'ckpts'
    env=build_environment(cfg)
    agent = SAC(env.observation_space.shape[0], env.action_space, gamma=gamma, tau=tau, alpha=alpha, policy=policy,
        target_update_interval=target_update_interval, automatic_entropy_tuning=automatic_entropy_tuning,
        hidden_size=hidden_size, learning_rate=learning_rate, task_name=env_name, seed=seed)
    
    for i in range(episodes):
        task_done = False
        agent.load_only_policy_checkpoint(env_name, checkpoint_dir, evaluate=True, seed = cfg.seed, suffix='100')    
        state = env.reset()
        episode_reward = 0
        done = False
        while not done:
            action = agent.select_action(state, evaluate=True)
            next_state, reward, done, info = env.step(action)  # Step
            # done = done
            env.render()
            episode_reward += reward
            state = next_state
            # Add a delay to slow down the simulation
            time.sleep(delay)

            if(info['success'] == 1.0): 
                task_done = True

        print(f"Episode {i} finished with a reward of {episode_reward}")
        print('Tash success : {}'.format(task_done))

reginald-mclean, 2024-08-26:
I can't suggest a solution without the code used in `build_environment`

jayaramreddy10, 2024-08-27:
Sorry, forgot to mention about that. 

from gym.wrappers import TimeLimit

def build_environment(cfg, render=False):
          env_name = cfg.env_name  # 'push-v2-goal-observable'
          env = metaworld.envs.ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE[env_name](seed=seed)
          env = MetaWorldGymWrapper(env)  # gym wrapper
          env = TimeLimit(env, max_episode_steps=episode_length)

class MetaWorldWrapper(gym.Wrapper):
	def __init__(self, env):
		gym.Wrapper.__init__(self, env)
	def reset(self):
		state = self.env.reset()
		return state
	def step(self, action):
		next_state, reward, done, info = self.env.step(action)
		return next_state, reward, done, info
	def render(self, mode=None, height=384, width=384, camera_id=None):
		self.env.camera_id=camera_id
		return self.env.render()

reginald-mclean, 2024-08-27:
What commit of Metaworld are you using?

jayaramreddy10, 2024-08-27:
I am currently using v0.0.0.

reginald-mclean, 2024-08-27:
How did you install that? 
If you follow the installation instructions on the readme you should be installing something that doesn't have the issue you mentioned

jayaramreddy10, 2024-08-27:
I had some issues when trying with latest metaworld (v2.0.0) as I was using older gym 0.23.0. So, I had installed it from one of the baseline repo for our project (https://github.com/XuGW-Kevin/DrM/tree/main/metaworld).  Is there a way to install metaworld version compatible with gym 0.23.0?

reginald-mclean, 2024-08-27:
`pip install git+https://github.com/Farama-Foundation/Metaworld.git@04be337a12305e393c0caf0cbf5ec7755c7c8feb`

This command installs the pre-Farama Foundation Meta-World. I don't know exactly which Meta-World  version would be usable with gym 0.23.0. I think it's missing a required package install of scipy. You're currently using a version of Meta-World that I can't provide much help on as it's not the version we provide, it's from someone else's repo. If you use a version/commit of Meta-World from the Meta-World repo, then I can provide more support

jayaramreddy10, 2024-08-27:
Thanks, have just tried it.This version gives a stable state after repeated reset.