Farama-Foundation/Metaworld #505: Wrong camera view when using "gripper_pov" view to render images

Aoyama7Hai, 2024-09-12:
I am using metaworld (pip show version = 2.0.0) try to get different camera views as shown : 
[image]
but is seems like the camera with camera_name='gripper_pov' or camera_id='5' renders in a wrong view

Should this view seems like this one (figure 1.a  in other papers) ?
[image]
How to solve this problem? I don't know the causes of this problem, thank you !

Aoyama7Hai, 2024-09-13:
By the way, when I try to set 'caemra_id' = 4 , the rendering image like this:
[image]
It seems like the image is upside down, maybe this clue can be useful to figure out the question.

Aoyama7Hai, 2024-10-31:
Hello? I wonder if there are some users or developers who have faced reletive experiences to help to fix this problem?
If this is relavant to the env config, Ican provide my setting details to together fix this bug.

Kallinteris-Andreas, 2024-11-02:
the camera upside down might be a windows thing @Aoyama7Hai what OS are you using

Aoyama7Hai, 2024-11-02:
@Kallinteris-Andreas I'm using Ubuntu 22.04.4 LTS (GNU/Linux 6.5.0-44-generic x86_64)
BTW, I wonder the the render image of camera_name='gripper_pov' a.k.a camera_id='5' in the first conmmet is right? It seems like the position of the camera were set up incorrectly inside the gripper

alexcbb, 2025-03-06:
Hello @Aoyama7Hai I proposed a pull request to fix the issue you proposed : https://github.com/Farama-Foundation/Metaworld/pull/525 tell me if it answers your question

closed 2025-03-17 by reginald-mclean