Farama-Foundation/Metaworld #513: Multiple cameras in render

AmpiroMax, 2024-10-18:
Is there any way to render image from several cameras in current env?

```python
ml1 = metaworld.ML1(env_name) 
cameras_name = ['topview', 'corner'] 
env = ml1.train_classes[env_name](render_mode=render_mode, camera_name=cameras_name)
```

reginald-mclean, 2024-10-24:
If you're using the most recent version of Meta-World, you should be able to specify the camera_id or camera_name. The render function uses [this function from Gymnasium](https://github.com/Farama-Foundation/Gymnasium/blob/dc36e9654338ef4b9cc453ba64b2dee7473dc593/gymnasium/envs/mujoco/mujoco_rendering.py#L204)

Kallinteris-Andreas, 2024-10-24:
Currently we do not support multi-camera rendering., You could try to fake it by using the render function and then changing the internal `camera_id` Member variable of the MuJoCo renderer

samb271, 2025-08-28:
Hi! I'm trying to get multiple camera views from a single environment, but am having a hard time. Did you get this to work? If so, would you mind sharing a tiny code snippet to help get started? Any help would be much appreciated!