Farama-Foundation/Metaworld #520: AttributeError: 'AssemblyV2GoalObservable' object has no attribute 'sim'

Yushi-Du, 2024-12-23:
I was trying to run the drawer-open-v2 environment while I got the error below:

Traceback (most recent call last):
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/hydra_launcher.py", line 38, in <module>
    configure_jobs()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/main.py", line 48, in decorated_main
    _run_hydra(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 377, in _run_hydra
    run_and_report(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 294, in run_and_report
    raise ex
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 211, in run_and_report
    return func()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 378, in <lambda>
    lambda: hydra.run(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 111, in run
    _ = ret.return_value
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/core/utils.py", line 233, in return_value
    raise self._return_value
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/core/utils.py", line 160, in run_job
    ret.return_value = task_function(task_cfg)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/hydra_launcher.py", line 34, in configure_jobs
    bc_train_loop(job_data)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/train_loop.py", line 151, in bc_train_loop
    paths = sample_paths(num_traj=job_data['eval_num_traj'], env=e, #env_constructor,
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/sampling.py", line 172, in sample_paths
    return do_rollout(**input_dict)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/sampling.py", line 95, in do_rollout
    o = env.reset()
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/gym_env.py", line 91, in reset
    return self.env.reset()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 279, in reset
    observation = self.env.reset(**kwargs)
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 280, in reset
    return self.observation(observation)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/obs_wrappers.py", line 399, in observation
    return self.get_image()
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/obs_wrappers.py", line 391, in get_image
    img = self.sim.render(width=self.width, height=self.height, depth=self.depth,
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 238, in __getattr__
    return getattr(self.env, name)

I wonder if it has something to do with mujoco versions? I would really appreciate your reply and help. Thank you very much.

Yushi-Du, 2024-12-23:
This is the entire output:

Traceback (most recent call last):
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/hydra_launcher.py", line 38, in <module>
    configure_jobs()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/main.py", line 48, in decorated_main
    _run_hydra(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 377, in _run_hydra
    run_and_report(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 294, in run_and_report
    raise ex
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 211, in run_and_report
    return func()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 378, in <lambda>
    lambda: hydra.run(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 111, in run
    _ = ret.return_value
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/core/utils.py", line 233, in return_value
    raise self._return_value
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/core/utils.py", line 160, in run_job
    ret.return_value = task_function(task_cfg)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/hydra_launcher.py", line 34, in configure_jobs
    bc_train_loop(job_data)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/train_loop.py", line 151, in bc_train_loop
    paths = sample_paths(num_traj=job_data['eval_num_traj'], env=e, #env_constructor,
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/sampling.py", line 172, in sample_paths
    return do_rollout(**input_dict)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/sampling.py", line 95, in do_rollout
    o = env.reset()
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/gym_env.py", line 91, in reset
    return self.env.reset()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 279, in reset
    observation = self.env.reset(**kwargs)
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 280, in reset
    return self.observation(observation)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/obs_wrappers.py", line 399, in observation
    return self.get_image()
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/obs_wrappers.py", line 391, in get_image
    img = self.sim.render(width=self.width, height=self.height, depth=self.depth,
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 238, in __getattr__
    return getattr(self.env, name)
AttributeError: 'AssemblyV2GoalObservable' object has no attribute

Yushi-Du, 2024-12-23:
Traceback (most recent call last):
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/hydra_launcher.py", line 38, in <module>
    configure_jobs()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/main.py", line 48, in decorated_main
    _run_hydra(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 377, in _run_hydra
    run_and_report(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 294, in run_and_report
    raise ex
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 211, in run_and_report
    return func()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 378, in <lambda>
    lambda: hydra.run(
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 111, in run
    _ = ret.return_value
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/core/utils.py", line 233, in return_value
    raise self._return_value
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/hydra/core/utils.py", line 160, in run_job
    ret.return_value = task_function(task_cfg)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/hydra_launcher.py", line 34, in configure_jobs
    bc_train_loop(job_data)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/train_loop.py", line 151, in bc_train_loop
    paths = sample_paths(num_traj=job_data['eval_num_traj'], env=e, #env_constructor,
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/sampling.py", line 172, in sample_paths
    return do_rollout(**input_dict)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/sampling.py", line 95, in do_rollout
    o = env.reset()
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/gym_env.py", line 91, in reset
    return self.env.reset()
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 279, in reset
    observation = self.env.reset(**kwargs)
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 280, in reset
    return self.observation(observation)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/obs_wrappers.py", line 399, in observation
    return self.get_image()
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/obs_wrappers.py", line 391, in get_image
    img = self.sim.render(width=self.width, height=self.height, depth=self.depth,
  File "/home/wangyx/miniconda3/envs/mujoco3/lib/python3.10/site-packages/gym/core.py", line 238, in __getattr__
    return getattr(self.env, name)
AttributeError: 'AssemblyV2GoalObservable' object has no attribute 'sim'

Kallinteris-Andreas, 2024-12-23:
What metaworld version are you running

reginald-mclean, 2024-12-23:
This looks like it is from R3M so you will need to install a specific commit of Meta-World that works with R3M. The last command in the [installation instructions](https://github.com/Farama-Foundation/Metaworld?tab=readme-ov-file#installation) should work. `pip install git+https://github.com/Farama-Foundation/Metaworld.git@04be337a12305e393c0caf0cbf5ec7755c7c8feb`

Yushi-Du, 2025-01-04:
Sorry for replying late, I was handling some other issues last week.

Thanks a lot! It seems that @reginald-mclean 's solution works for me. I am running metaworld for R3M.

> This looks like it is from R3M so you will need to install a specific commit of Meta-World that works with R3M. The last command in the [installation [...]

However, I ran into another issue which may have something to do with mujoco-py version: 

Traceback (most recent call last):
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/hydra_launcher.py", line 38, in <module>
    configure_jobs()
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/hydra/main.py", line 48, in decorated_main
    _run_hydra(
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/hydra/_internal/utils.py", line 377, in _run_hydra
    run_and_report(
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/hydra/_internal/utils.py", line 294, in run_and_report
    raise ex
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/hydra/_internal/utils.py", line 211, in run_and_report
    return func()
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/hydra/_internal/utils.py", line 378, in <lambda>
    lambda: hydra.run(
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 111, in run
    _ = ret.return_value
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/hydra/core/utils.py", line 233, in return_value
    raise self._return_value
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/hydra/core/utils.py", line 160, in run_job
    ret.return_value = task_function(task_cfg)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/hydra_launcher.py", line 34, in configure_jobs
    bc_train_loop(job_data)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/core/train_loop.py", line 151, in bc_train_loop
    paths = sample_paths(num_traj=job_data['eval_num_traj'], env=e, #env_constructor,
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/sampling.py", line 172, in sample_paths
    return do_rollout(**input_dict)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/sampling.py", line 95, in do_rollout
    o = env.reset()
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/gym_env.py", line 91, in reset
    return self.env.reset()
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/gym/core.py", line 278, in reset
    observation = self.env.reset(**kwargs)
  File "/home/wangyx/miniconda3/envs/mujoco5/lib/python3.10/site-packages/gym/core.py", line 279, in reset
    return self.observation(observation)
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/obs_wrappers.py", line 399, in observation
    return self.get_image()
  File "/home/wangyx/r3m-eval/evaluation/r3meval/utils/obs_wrappers.py", line 391, in get_image
    img = self.sim.render(width=self.width, height=self.height, depth=self.depth,
  File "mjsim.pyx", line 153, in mujoco_py.cymj.MjSim.render
  File "wrappers.pxi", line 1071, in mujoco_py.cymj.PyMjModel.camera_name2id
ValueError: No "camera" with name left_cap2 exists. Available "camera" names = ('topview', 'corner', 'corner2', 'corner3', 'behindGripper', 'gripperPOV').

I have tried to install different mujoco-py versions including 2.1.2.14 and 2.0.2.8 but both failed in my case. I would like to ask if you have encountered similar problems？Thank you very much for the reply.

Kallinteris-Andreas, 2025-01-04:
Try passing the camera argument during environment construstion

reginald-mclean, 2025-01-04:
R3M uses a fork of Meta-World that is different from our repo, they add additional cameras as seen here:
https://github.com/suraj-nair-1/metaworld/commit/4e896a5de0549fbc0b57ce9e7f86d8c98ca8662b#diff-c8eec7ed8050fcecdab0f09da27c0236a4ab1abfa80b75b6494096f6c228648b

You can try adding the camera into the relevant locations in the Meta-World version you have or use the fork of Meta-World that R3M uses

closed 2025-01-04 by reginald-mclean