Farama-Foundation/Metaworld #545: 'ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE' is missing

ale4638, 2025-06-26:
### Question

I'm tring to import  lift3d.envs from MetaWorldEnv. However, the issue occurred. 

(lift3d) [issacb25@SPGL-1-1 LIFT3D]$ python -m lift3d.tools.sample_obs_metaworld
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/grp01/cs_hszhao/ale/LIFT3D/lift3d/tools/sample_obs_metaworld.py", line 8, in <module>
    from lift3d.envs import MetaWorldEnv
  File "/grp01/cs_hszhao/ale/LIFT3D/lift3d/envs/__init__.py", line 2, in <module>
    from .metaworld_env import (
  File "/grp01/cs_hszhao/ale/LIFT3D/lift3d/envs/metaworld_env.py", line 10, in <module>
    from metaworld.envs import ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE
ImportError: cannot import name 'ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE' from 'metaworld.envs' (/home/issacb25/.conda/envs/lift3d/lib/python3.11/site-packages/metaworld/envs/__init__.py)

ale4638, 2025-06-26:
I've checked metaworld.env and there is no any variable named 'ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE' .

reginald-mclean, 2025-07-11:
You won't find a variable like that because the current version of MetaWorld is V3.

closed 2025-07-12 by reginald-mclean